platformio/platformio-registry #3: Claiming viralinkio/ViraLink-MQTT-Client

armp30, 2022-01-08:
#### Package information

- Package link: https://registry.platformio.org/libraries/viralinkio/ViraLink-MQTT-Client

#### Description

We are maintaining this package under the [viralinkio](https://github.com/viralinkio) organization on github.

ivankravets, 2022-01-08:
Do you have any issues with access to this package?

armp30, 2022-01-08:
the release versions are automatic and i dont have any control on it

ivankravets, 2022-01-08:
Please login to the PlatformIO using Github. See https://docs.platformio.org/en/latest/plus/pio-account.html 

Please inform me when you created `armp30` account on PlatformIO. We will make you an owner of `viralinkio` org.

armp30, 2022-01-08:
i created armp30 account

armp30, 2022-01-08:
is automatic deployment based on version number change on library.json file?

armp30, 2022-01-08:
@valeros

armp30, 2022-01-08:
@ivankravets

ivankravets, 2022-01-08:
You are the owner of https://registry.platformio.org/search?q=owner%3Aviralinkio now. Please use `pio package publish --owner viralinkio` next time.

closed 2022-01-08 by ivankravets