platformio/platformio-registry #5: Claiming namespace mannypeterson/HeliOS

MannyPeterson, 2022-01-21:
#### Package information

- Package link: https://registry.platformio.org/libraries/mannypeterson/HeliOS

#### Description

ivankravets, 2022-01-21:
Hi @MannyPeterson,

Have you tried to log in to [PlatformIO Account\(https://docs.platformio.org/en/latest/plus/pio-account.html) using your Github account?

MannyPeterson, 2022-01-22:
@ivankravets I have but the very email step does not seem to work for me, I never receive the verification email and I have checked the spam folder.

Also, the package seems to still only show release 0.2.6 even though I released 0.2.7 a few days ago.

MannyPeterson, 2022-01-23:
Hi @ivankravets 

Thank you for fixing the crawler. Everything looks good now. 😊

The email address you provided is old but it still works and I didn't get the verification email. I did remove that email address from my GitHub account. I will try again.

❤️🇺🇦

MannyPeterson, 2022-01-24:
Okay, all good now. I logged in with GitHub. What’s next?

ivankravets, 2022-01-24:
I think nothing :)
If you prefer manually publish updates for your library, you can now use https://docs.platformio.org/en/latest/core/userguide/package/cmd_publish.html

closed 2022-01-24 by ivankravets