platformio/platformio-registry #24: [Error] NotFoundError: Failed to execute 'removeChild' on 'Node': The node to be removed is not a child of this node.

JoaoBarioni, 2022-07-15:
# Page
https://registry.platformio.org/libraries/mbed-jackb/EEPROM/installation

# Stack
```
at O (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:456996)
    at Suspense
    at div
    at div
    at Q (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:152703)
    at div
    at main
    at X (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:155477)
    at F (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:456370)
    at R (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:611100)
    at O (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:456996)
    at B (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:456426)
    at b (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:447917)
    at https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:562190
    at hl (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:411438)
    at hl (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:411438)
```

ivankravets, 2022-07-30:
Please refresh the page with Ctrl+R. Does it work now?

closed 2022-07-30 by ivankravets

JoaoBarioni, 2022-07-30:
Yes, I think this error was momentary, thanks for your reply.