platformio/platformio-registry #25: The registry has two identical M5Unit-Sonic

Tinyu-Zhao, 2022-07-29:
If you [search](https://registry.platformio.org/search?q=M5Unit-Sonic) for M5Unit-Sonic in the registry, you will get two results and no response when you click on it, other [repositories](https://registry.platformio.org/libraries/m5stack/M5Unit-Sonic) do not have this problem

[image]

[image]

Tinyu-Zhao, 2022-07-29:
I just found another problem, if I use the Github login on Platform IO in VsCode, I get a 404.
The problem is similar to [this one](https://community.platformio.org/t/ide-bug-vscode-login-with-social/28984/2)

https://user-images.githubusercontent.com/44396641/181695537-c951e5aa-bf59-4a69-945f-82ecdb5988ed.mov

PlatformIO IDE Version v2.5.1
PlatformIO Core, version 6.1.4a2
VsCode Version 1.69.2

ivankravets, 2022-07-29:
Thanks for the report, fixed both issues. See https://community.platformio.org/t/ide-bug-vscode-login-with-social/28984/8

closed 2022-07-29 by ivankravets