platformio/platformio-registry #30: Unable to remove package

dyarkovoy, 2022-08-05:
`pio pkg unpublish "dyarkovoy/ESP32 logger@1.0.2"`

returns with

`Error: You can unpublish a package version only in 72 hours since the publish date 2021-02-14 16:20:25`

IMHO, it has been well over 72 hours since 2021-02-14

ivankravets, 2022-08-05:
and how could we help here?

dyarkovoy, 2022-08-05:
Please help by letting me know how I can remove my package `dyarkovoy/ESP32 logger@1.0.2`

ivankravets, 2022-08-05:
Do you plan to publish a new version?

dyarkovoy, 2022-08-05:
This package was published by mistake, so I need to remove all packages under the `dyarkovoy` namespace. The correct namespace is `esp32m` and the correct package is already published here: https://registry.platformio.org/libraries/esp32m/ESP32%20logger

ivankravets, 2022-08-05:
Done! Thanks for using PlatformIO!

closed 2022-08-05 by ivankravets