platformio/platformio-registry #31: Claiming namespace mdkendall

mdkendall, 2022-09-27:
#### Package information

- https://registry.platformio.org/libraries/mdkendall/Thingpings

#### Description

Two weeks ago I created the ThingpingsLib repo on GitHub https://github.com/mdkendall/ThingpingsLib, created release 1.0.0, and published it in the Arduino Library Manager here https://www.arduino.cc/reference/en/libraries/thingpings/

This had the side effect of the library appearing in the PlatformIO registry as above.

A few days later I created release 1.1.0 on GitHub. This was automatically picked up by the Arduino Library Manager. But it was not picked up by PlatformIO which still lists 1.0.0 as the latest. I do not know why.

I thought I might try using the `pio pkg publish` command to force the PlatformIO registry to update. I do not have a PlatformIO account, so I attempted to login using the "Login with GitHub" OAuth option. I successfully authorised this on GtHub but the PlatformIO login fails with "We are sorry. Account is disabled, contact your administrator." I don't know why that is and I don't have an "administrator".

What I would like is:
- to be able to login to PlatformIO with my GitHub account
- for that account to be the owner of the above library 

I would also like to understand why the PlatformIO registry is not picking up the 1.1.0 release automatically, so I can avoid this problem in the future.

ivankravets, 2022-09-30:
Thanks for contacting us! Could you try again to login with Github? Does it work now?

mdkendall, 2022-09-30:
Thank you for your help.

If I try to log in with GitHub now I see a "Account already exists" message.

[image]

If I click "Add to existing account" I see a message that says a verification code has been sent by email, but I never receive any email, even if I click to re-send it.

[image]

ivankravets, 2022-09-30:
I see. The account does not contain email record. Could you send us your email linked with GitHub to contact@platformio.org. Thanks!

mdkendall, 2022-09-30:
I am now able to log in and I see that I am the owner of the Thingpings library. Thank you very much for your help.

closed 2022-09-30 by mdkendall

ivankravets, 2022-10-01:
Great! Thanks for using PlatformIO 🙏