platformio/platformio-registry #35: Claiming namespace [gitlab-thijsvanulden/PowerNap]

Thijxx, 2022-10-27:
#### Package information

- Package link: https://registry.platformio.org/libraries/gitlab-thijsvanulden/PowerNap

#### Description

It has been a couple of years since I added this to the registry and I don't know how I did that back then.
There seems to be no account management on the registry, no 'update package' in `pio pkg` and when I try to publish the latest version of my library I get the following.

[image]

I probably need to aquiere owenership again, my mail and account on platformio is thijsvanulden AT gmail
Thanks!

ivankravets, 2022-10-27:
Done! https://registry.platformio.org/libraries/thijxx/PowerNap

Thanks for using PlatformIO!

closed 2022-10-27 by ivankravets

Thijxx, 2022-10-27:
Could you explain what has changed? I cannot see anything new using `pio access list` and when I publish the new version I still get the warning `Warning! A package with the name `PowerNap` is already published by the next owners: gitlab-thijsvanulden`

Thijxx, 2022-10-27:
nevermind, it just took a couple of minutes to process I guess, thank for the quick response!