platformio/platformio-registry #37: [Error] TypeError: Invalid Version: null

luisr320, 2022-11-04:
# Page
https://registry.platformio.org/libraries/raronoff/clickButton/versions

# Stack
```
at q (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:150318)
    at O (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:456996)
    at Suspense
    at div
    at div
    at Q (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:152703)
    at div
    at main
    at X (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:155477)
    at F (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:456370)
    at R (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:611100)
    at O (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:456996)
    at B (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:456426)
    at b (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:447917)
    at https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:562190
    at L (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:609966)
    at I (https://cdn.registry.platformio.org/static/js/main.7cce2b10.js:2:611294)
```

luisr320, 2022-11-30:
Still gives me the same error

ivankravets, 2022-11-30:
Thanks! The package was removed from the registry.

luisr320, 2022-11-30:
Ok, it's showing a 404 now. Shal I close this comment?

ivankravets, 2022-11-30:
Yes, the package was removed by its owner. Please search in the registry for alternative packages.