platformio/platformio-registry #39: Release notes link to tag name that doesn't exist

nomis, 2022-11-06:
The release notes on the website link to tag `v${version}` which doesn't exist for my libraries because the tag is `${version}`.

The documentation for `library.json` implies that the tag name should be `${version}`, not `v${version}` because that is the canonical location of the source code:

https://docs.platformio.org/en/latest/manifests/library-json/fields/repository.html:
> `branch` if is not specified, default branch will be used. This field will be ignored if tag/release exists with the value of [version](https://docs.platformio.org/en/latest/manifests/library-json/fields/version.html)

There are also many libraries that only have `v${version}` as a tag but haven't explicitly specified this in `library.json`. I assume the historical auto-publish lookup behaviour actually checked for both variants, but the registry website doesn't do this?

ivankravets, 2022-11-15:
Thanks for the report. We will think about how to fix it on the frontend side.

ajsb85, 2023-01-04:
https://docs.github.com/es/rest/releases/releases

```console
curl \
  -H "Accept: application/vnd.github+json" \
  -H "Authorization: Bearer <YOUR-TOKEN>"\
  -H "X-GitHub-Api-Version: 2022-11-28" \
  https://api.github.com/repos/OWNER/REPO/releases
```