platformio/platformio-registry #52: GitHub Repository Location Change for Registry Package [HeliOS]

MannyPeterson, 2023-03-21:
@ivankravets @valeros 

I was not sure where to open this issue, I hope this is acceptable. I am the owner/maintainer of the "HeliOS" package (see Registry link below). The GitHub repository has moved from https://github.com/mannypeterson/HeliOS to https://github.com/heliosproj/HeliOS. I have updated the "library.json" file. How do I get the Registry to reference the new GitHub repository location? I do have a PIO login. Do I just upload a new package? Please advise.

[https://registry.platformio.org/libraries/mannypeterson/HeliOS](https://registry.platformio.org/libraries/mannypeterson/HeliOS)

Thank you for PlatformIO and God bless Ukraine,

Manny

ivankravets, 2023-03-22:
Hi @MannyPeterson,

Thanks for using PlatformIO!

You don't need to apply any steps on your side. We created the `heliosproj` organization and made you an owner of it. Also, the library `HeliOS` has been moved to it. 

You can list organizations with the `pio org list` command. Next time when you will publish the new version of `HeliOS`, please use the following command:

```
pio pkg publish --owner heliosproj
```

See https://registry.platformio.org/libraries/heliosproj/HeliOS

closed 2023-03-22 by ivankravets

MannyPeterson, 2023-03-22:
Thank you @ivankravets !!