platformio/platformio-registry #54: Publishing package reverts package back to old repo location

MannyPeterson, 2023-03-29:
PIO Friends,

The library "HeliOS" was recently moved from my person GH account mannypeterson/HeliOS to a GH org heliosproj/HeliOS. You graciously updated the package in the registry to the new location (thank you). But I published a new package (0.4.1) and it reverts to mannypeterson/HeliOS even though the contents of the library.json are correct. So now there are two packages in the registry.

[https://registry.platformio.org/libraries/heliosproj/HeliOS](https://registry.platformio.org/libraries/heliosproj/HeliOS) <-- Contains 0.4.0

[https://registry.platformio.org/libraries/mannypeterson/HeliOS](https://registry.platformio.org/libraries/mannypeterson/HeliOS) <-- Contains 0.4.1

[https://raw.githubusercontent.com/heliosproj/HeliOS/master/library.json](https://raw.githubusercontent.com/heliosproj/HeliOS/master/library.json) <-- library.json file

What am I doing wrong? I appreciate the help!!

Manny

ivankravets, 2023-03-29:
How did you publish the library? Have you used `pio pkg publish --owner heliosproj`? It seems you missed the `owner` option. As result, the library was published from your personal account

closed 2023-03-29 by ivankravets

MannyPeterson, 2023-03-29:
@ivankravets 

That fixed it. Thank you!