platformio/platformio-registry #62: [Error] TypeError: e is undefined

JackYang-hellobobo, 2023-05-07:
# Page
https://registry.platformio.org/platforms/platformio/espressif32/examples/arduino-blink/lib

# Stack
```
xe@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:72757
b@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:155242
div
div
D@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:162203
_@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:481897
T@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:484521
Suspense
div
ee@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:189970
div
main
ne@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:192862
_@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:481897
F@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:483755
P@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:641020
_@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:481897
T@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:484521
B@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:484004
c@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:475207
1010/J<@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:591849
R@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:639886
I@https://cdn.registry.platformio.org/static/js/main.c2f4f14c.js:2:641214
```

ivankravets, 2023-05-07:
Please press Ctrl+R. Should be fixed.

closed 2023-05-07 by ivankravets

JackYang-hellobobo, 2023-05-07:
Thank U ~