platformio/platformio-registry #66: Why are packages being removed/disappearing from the Registry?

kuba2k2, 2023-06-22:
Hi

I'm the owner of [LibreTiny platform](https://registry.platformio.org/platforms/kuba2k2/libretiny/versions) package on the PIO registry.
The project was formerly called LibreTuya, and such package was also published in the registry.

What I don't understand is, why is the previous package gone, even though I didn't request it to be unpublished?
Also, the new package currently has only one version available (1.1.0), even though versions 1.0.1 and 1.0.2 have been published and available before. Again, I didn't request them to be removed. The same issue was happening to the previous LibreTuya platform (old versions were disappearing).

Is there something wrong on my end? (i.e. some setting in the CI publishing workflow, or some account setting?)

closed 2023-06-22 by ivankravets

kuba2k2, 2023-06-22:
Hi, what runtime API and dependencies do you mean? Were the versions removed manually by the administration team, or is it an automated process? Can I request all versions to be kept for compatibility purposes in some way? (projects might require a specific version of the platform)

ivankravets, 2023-06-23:
> what runtime API and dependencies do you mean?
https://github.com/kuba2k2/libretiny/blob/master/platform.py

> Were the versions removed manually by the administration team

Yes, we restored them.

> Can I request all versions to be kept for compatibility purposes in some way? 

We will not touch further release if this is the final stable development platform. We saw duplicated platforms from your account before. Even the current version looks like an experimental demo. The link is broken to the repository https://registry.platformio.org/platforms/kuba2k2/libretiny

kuba2k2, 2023-06-23:
The duplicated platform was because of the project rename (which was necessary due to possible copyright issues). That was done along with a bigger refactor, so I don't need the old platform to be available anymore.

The current version is stable, working, and used by many people on hundreds of devices. I'm not sure what makes it look like an experimental demo; the broken repository link is just a little thing that I've missed (it's only visible on PIO registry anyway). The fact that it evolves quickly doesn't necessarily mean that current versions aren't stable.

The project name is final, but that also doesn't mean the platform will not get newer versions than 1.1.0 - it certainly will. Even then, keeping old versions is just a good practice, and shouldn't mean that any released version makes old versions unstable.

To summarize, I'd like to request all previous (and future) versions of LibreTiny not to be removed from the registry, because it may be very problematic and break projects like ESPHome (which depend on a specific version). Let me know if that's possible, thanks.

ivankravets, 2023-06-23:
Thanks for the updates! No new versions will be deleted in the future.

>  break projects like ESPHome (which depend on a specific version). 

Could you point me to the source of code on which version ESPHome depends?

kuba2k2, 2023-06-23:
Here:
https://github.com/kuba2k2/libretiny-esphome/blob/platform/libretiny-prerelease/esphome/components/libretiny/__init__.py#L159

ESPHome support hasn't been merged into upstream yet, but we're working with the ESPHome team towards doing that soon. The same version constraint is there for esp32 or esp8266 support.