platformio/platformio-registry #78: Library tobiasschuerg/MH-Z-CO2-Sensors doesn't automatically updating from Arduino library registry

SNMetamorph, 2023-10-27:
Latest available version of library on [Platfrom.io registry](https://registry.platformio.org/libraries/tobiasschuerg/MH-Z%20CO2%20Sensors): 1.4.0
Actual available versions of this library: https://github.com/tobiasschuerg/MH-Z-CO2-Sensors/tags

ivankravets, 2023-10-28:
Sorry, we can't fix this. The owner (@tobiasschuerg) changed the library name from `MH-Z CO2 Sensors` to `MH-Z CO2 Sensors (MH-Z14A, MH-Z14B, MH-Z19A,MH-Z19B, MH-Z19C)` which is totally different package according to the declaration rules.

Once the library changes its name to `MH-Z CO2 Sensors` it will be automatically updated.

@tobiasschuerg , could you move ` (MH-Z14A, MH-Z14B, MH-Z19A,MH-Z19B, MH-Z19C)` to the `description` field?

tobiasschuerg, 2023-11-01:
I wasn't aware that this would break the something (wouldn't it be better to use the repository as package? or just manually set a package name?).

The new version `1.5.1` is now using the old package name:
https://github.com/tobiasschuerg/MH-Z-CO2-Sensors/releases/tag/1.5.1

closed 2023-11-01 by SNMetamorph

ivankravets, 2023-11-02:
Thanks, @tobiasschuerg ! https://registry.platformio.org/libraries/tobiasschuerg/MH-Z%20CO2%20Sensors