platformio/platformio-registry #83: Claiming namespace marlinfirmware

thinkyhead, 2023-12-27:
#### Package information

- Package link: https://registry.platformio.org/libraries/marlinfirmware/U8glib-HAL

#### Description

The namespace doesn't seem to be associated with my PIO account ("thinkyhead") so I'm denied when I try to publish updates to packages under that namespace. Is that namespace currently unclaimed?

ivankravets, 2023-12-28:
Hi Scott,

Thanks for contacting us. You are now the owner of `marlinfirmware` organization. Please don't forget to use `pio pkg publish --owner marlinfirmware` when publishing a new version or package.

Happy holidays!

closed 2023-12-28 by ivankravets

thinkyhead, 2023-12-29:
Thank you Ivan. Happy holidays to you, and best wishes for the new year!