platformio/platformio-registry #85: Claiming namespace wolfSSL

gojimmypi, 2024-01-06:
#### Package information

- Package link: https://registry.platformio.org/libraries/onelife/wolfssl

#### Description

We at [wolfSSL](https://www.wolfssl.com/) have neglected to setup an official registry library at PlatformIO. This has resulted in others such as `onelife` to publish their own copy.

Although instances like this are appreciated for the enthusiasm and effort, there's no long term support nor assurance that the library is properly installed or even genuine wolfSSL source. This leads to less than exceptional first impressions for both PlatformIO and wolfSSL users. The current library as-is will not even compile. It is also old and missing many updated security and usability features.

I'd like to fix that.

This is partly in response to a [recent wolfSSL forum issue](https://www.wolfssl.com/forums/post7413.html) where a new user encountered significant challenges when attempting to use the [onelife/wolfssl library](https://registry.platformio.org/libraries/onelife/wolfssl). I've documented the [steps needed](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix/IDE/ARDUINO/PlatformIO) to repair and configure the current PlatformIO wolfSSL library. However, I'd like to instead have an official registry.

Please let me know what I can do to publish the official wolfSSL libraries on PlatformIO. 

Thank you

Jim
wolfSSL Software Engineer
jim (at) wolfssl.com

See the Espressif [live](https://components.espressif.com/components?q=namespace:wolfssl) and [staging](https://components-staging.espressif.com/components?q=namespace:gojimmypi) sites for our ESP Registry Managed Components.

ivankravets, 2024-01-06:
I think we should follow the following steps:

1. Create `wolfSSL` org using https://docs.platformio.org/en/latest/core/userguide/org/index.html
2. Publish a new working version using https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_publish.html  and `--owner option`. Example, `pio pkg publish --owner wolfSSL path/to/lib/folder`
3. Would you like to remove unofficial and broken https://registry.platformio.org/libraries/onelife/wolfssl ? We can do this from our side according to the wolfSSL copyrights

A few hints:

* Creating library docs https://docs.platformio.org/en/latest/librarymanager/creating.html
* Advanced library build control https://docs.platformio.org/en/latest/manifests/library-json/fields/build/index.html , see also examples https://docs.platformio.org/en/latest/manifests/library-json/fields/build/builder.html#examples
* https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_pack.html is useful to generate a tar.gz package and see what will be published to the registry

gojimmypi, 2024-01-09:
Hi @ivankravets - That sounds great, thanks!

I tried logging in with my `gojimmypi` account, but I am unable to created an org. 

```
$ /home/gojimmypi/.platformio/penv/bin/platformio org add gojimmypi gojimmypi
AccountError: You do not have permission to add a new owner to organization 'gojimmypi'
```

So I created a new [wolfssl](https://community.platformio.org/u/wolfssl/) user:

```
$ /home/gojimmypi/.platformio/penv/bin/platformio account login --username wolfssl --password (mypassword)
Successfully logged in!
gojimmypi:~
$ /home/gojimmypi/.platformio/penv/bin/platformio org add wolfssl gojimmypi
AccountError: You do not have permission to add a new owner to organization 'wolfssl'
gojimmypi:~
$ /home/gojimmypi/.platformio/penv/bin/platformio org list
You do not have any organization
gojimmypi:~
$ /home/gojimmypi/.platformio/penv/bin/platformio org add wolfssl wolfssl
AccountError: You do not have permission to add a new owner to organization 'wolfssl'
```

I'd like to have a personal org for testing, using my personal ID `gojimmypi` , or perhaps something like or `wolfSSL-staging`, and an official org for publishing, named `wolfSSL` as you suggested. It might be nice to have an official, test/preview library. 

As you can tell, I'm completely new to PlatformIO. Thank you for your help.

ivankravets, 2024-02-28:
Please reopen if you still need help

closed 2024-02-28 by ivankravets

gojimmypi, 2024-03-09:
Hi @ivankravets and thank you for the publishing tips!

With the merge of https://github.com/wolfSSL/Arduino-wolfSSL/pull/1 and then a later polish with https://github.com/wolfSSL/Arduino-wolfSSL/pull/5 I have the first Official wolfSSL published here:

https://registry.platformio.org/libraries/wolfssl/wolfssl

The Arduino Library publish is here:

https://www.arduino.cc/reference/en/libraries/wolfssl/

As I am somewhat new to PlatformIO, I'd appreciate any review & suggestions for improvements.

ivankravets, 2024-03-09:
Hi, thanks for pinging me. I'm so glad to talk about this.

Comment by @dgarske  https://github.com/wolfSSL/Arduino-wolfSSL/pull/1#issuecomment-1944710160

> did you figure out a plan for how to avoid keeping duplicate files?

All these issues with duplicate files and restructuring the original WolfSSL repository are caused by the limitations of proprietary Arduino build systems. I would not recommend translating "dances with a tambourine" to the PlatformIO.

PlatformIO provides fine-grade access to the build process (runtime) and library maintainers have full control over it. There is [build](https://docs.platformio.org/en/latest/manifests/library-json/fields/build/index.html) field in `library.json` and you can configure the build flow at runtime (when the user builds a project). It allows library authors to keep source code without changes and adapt it to multiple frameworks including Arduino. There are a few examples https://docs.platformio.org/en/latest/manifests/library-json/fields/build/extrascript.html

Benefits from this approach:
1. We keep original source code of WolfSSL ( https://github.com/wolfSSL/wolfssl ) in the PlatformIO Registry
2. The library can be compatible with multiple frameworks
3. Easy to maintain, the [library.json](https://docs.platformio.org/en/latest/manifests/library-json/index.html) manifest can be added to the WolfSSL repository with its [extraScript](https://docs.platformio.org/en/latest/manifests/library-json/fields/build/extrascript.html) (similar to Cmake script) and configured for CI/CD
4. PlatformIO is very popular among Espressif customers, a lot of popular commercial and OSS projects are based on it. It is great for you to attract new customers for WolfSSL.

----

A few cents to https://github.com/wolfSSL/Arduino-wolfSSL . I would recommend to rename that library to `Arduino-wolfSSL` in https://github.com/wolfSSL/Arduino-wolfSSL/blob/main/library.properties#L1 . Otherwise, there is a conflict between two repositories/libraries having the same name but different source code. Once the `Arduino-wolfSSL` is added to the Arduino registry, it will be automatically synced with PlatformIO Registry, you don't need to apply extra steps.

gojimmypi, 2024-03-12:
@ivankravets thank you for the suggestions!

Some various comments as I encounter them:

# Initial Impression / Setup

The [VS Code documentation](https://docs.platformio.org/en/latest/integration/ide/vscode.html#ide-vscode) should probably be updated. It appears there's no longer a `PlatformIO Home` icon:

[image]

Instead, my VS Code icon is on the side, and looks like this:
 
[image]

My esp32dev install was a bit bumpy:

[image]

With errors like:

```
Library Manager: Warning! Package Mirror: [WinError 183] Cannot create a file when that file already exists: 'C:\\\\Users\\\\gojimmypi\\\\Documents\\\\PlatformIO\\\\Projects\\\\PatformIO_ESP32_wolfSSL\\\\.pio\\\\libdeps\\\\esp32dev\\\\wolfssl'\r\nLibrary Manager: Looking for another mirror...\r\nDownloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%\r\nUnpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%\r\n

Please manually remove the file `C:\\Users\\gojimmypi\\Documents\\PlatformIO\\Projects\\PatformIO_ESP32_wolfSSL\\.pio\\libdeps\\esp32dev\\wolfssl\\LICENSING`\r\n[WinError 5] Access is denied: 'C:\\\\Users\\\\gojimmypi\\\\Documents\\\\PlatformIO\\\\Projects\\\\PatformIO_ESP32_wolfSSL\\\\.pio\\\\libdeps\\\\esp32dev\\\\wolfssl\\\\README' 
```

# Library

> Would you like to remove unofficial and broken https://registry.platformio.org/libraries/onelife/wolfssl ? We can do this from our side according to the wolfSSL copyrights

Yes, please. Definitely. It appears the 3rd party, non-working library is listed first when searching for "wolfssl". If it was more obviously not an Official wolfSSL library, I might have left it. I need the supported wolfSSL to be clearly listed first. 

> dances with a tambourine

LOL, that's excellent. Indeed the Arduino formatting was quite an ordeal, and that's a fun description. Perhaps not a problem with single-file source libraries, but wolfSSL was considerably more challenging.

> I would recommend to rename that library

Ah yes, I see there are 2 different options for libraries: Arduino and Espidf.

[image]

Does that mean I need to publish two different libraries, one for each platform?

> Once the Arduino-wolfSSL is added to the Arduino registry, it will be automatically synced with PlatformIO Registry, you don't need to apply extra steps.

I didn't actually see that. My [Arduino publish](https://www.arduino.cc/reference/en/libraries/wolfssl/) (see [log](https://downloads.arduino.cc/libraries/logs/github.com/wolfSSL/Arduino-wolfSSL/)) does not appear to be visible in the PlatformIO library. The [wolfSSL PlatformIO Publish](https://registry.platformio.org/libraries/wolfssl/wolfssl) is based on the the ~~tambourine~~ Arduino-style library.

I appreciate all your suggestions. Cheers.

ivankravets, 2024-03-21:
> With errors like:

Hm, strange issue. Could you try to restart VSCode? Or, maybe, you already have fixed this issue?

> Yes, please. Definitely. It appears the 3rd party, non-working library i

Done! The library has been removed.

> I didn't actually see that

You will not see it due to the name conflicthttps://registry.platformio.org/libraries/wolfssl/wolfssl . 

Please fix the name of Arduino library https://github.com/wolfSSL/Arduino-wolfSSL/blob/main/library.properties . It should be 
```
name = Arduino-wolfSSL 
```

So, then we can have in the PlatformIO Registry 2 different libraries:
1. Original, with unmodified sources, the build process will be declared via `library.json -> build`. This library will work with any framework including Arduino
2. `Arduino-wolfSSL` - the Arduino's registry copy with a modified structure.

Please correct me if my vision is wrong.

gojimmypi, 2024-04-09:
Hi @ivankravets

Thank you for your suggestions and feedback.

> Done! The [3rd party unofficial wolfssl] library has been removed.

Confirmed! Thank you.

>>> Once the Arduino-wolfSSL is added to the Arduino registry, it will be automatically synced with PlatformIO Registry, you don't need to apply extra steps.

>> I didn't actually see that.

> You will not see it due to the name conflict https://registry.platformio.org/libraries/wolfssl/wolfssl .

I think I understand. I prefer to not change the Arduino library name. I'd like it be called simply "wolfssl" in the Arduino registry.

That said, I'm assuming that _only_ the _automated_ sync from Arduino libraries to PlatformIO will not work. Do I understand correctly? I was able to manually publish the Arduino one, as noted below and previously at https://registry.platformio.org/libraries/wolfssl/wolfssl

If that's the case, I'm ok with needing to manually publish 2 different libraries to PlatformIO with the respective names. It would be just one additional step in an automated process.

> Please fix the name of Arduino library `name = Arduino-wolfSSL`

Yes, I did that for a manual publish of the Arduino-wolfSSL library earlier today (see below).

> So, then we can have in the PlatformIO Registry 2 different libraries.

In order to test all this, I created a new `wolfssl-staging` org.

```
pio org create wolfssl-staging --email support@wolfssl.com --displayname "testing preview wolfssl"
```

See https://registry.platformio.org/search?q=owner%3Awolfssl-staging

In that org, I have two new wolfSSL libraries:

- Regular: https://registry.platformio.org/libraries/wolfssl-staging/wolfSSL
- Arduino: https://registry.platformio.org/libraries/wolfssl-staging/Arduino-wolfSSL

I edited the [library.properties](https://github.com/wolfSSL/Arduino-wolfSSL/blob/main/library.properties) for Arduino to be called `name=Arduino-wolfSSL` *only* for the PlatformIO Arduino version of wolfSSL.

The [Published wolfSSL Arduino Registry](https://www.arduino.cc/reference/en/libraries/wolfssl/) is still called simply `wolfssl` despite the restructured repo being called [Arduino-wolfSSL](https://github.com/wolfSSL/Arduino-wolfSSL).

After doing this, I noticed [your comment in Arduino-wolfSSL #7](https://github.com/wolfSSL/Arduino-wolfSSL/issues/7#issuecomment-2014758141):

> You can have only 1 library in the PlatformIO Registry which will be the mirror of https://github.com/wolfSSL/wolfssl. Having library.json manifest and platform-build.py extra script will let developers use original source code for any framework including Arduino.

The regular, non-Arduino library that I published to PlatformIO is not a copy of the [full wolfSSL repo](https://github.com/wolfSSL/wolfssl), but rather a subset of key directories for only the library.

That's pretty clever if PlatformIO can automatically restructure a codebase as complex as wolfSSL into something Arduino-library-friendly and still be operational for all the various target platforms supported by wolfSSL.

I will leave things as-is for now & do some testing to see just how this might work for wolfSSL apps. 

I'm interested in your feedback regarding having 2 _different_ libraries, as mentioned above - or perhaps only one single wolfSSL instance on PlatformIO that works for either environment.

For reference: published to PlatformIO registry like this:

```
C:\workspace\wolfssl-gojimmypi\IDE\PlatformIO>pio pkg publish --owner wolfssl-staging C:\workspace\Arduino-wolfSSL
Preparing a package...
Type:     library
Owner:    wolfssl-staging
Name:     Arduino-wolfSSL
Version:  5.6.6-Arduino.2
Size:     9.15MB
Are you sure you want to publish the library wolfssl-staging/Arduino-wolfSSL@5.6.6-Arduino.2 to the registry?
 [y/N]: y
The package publishing may take some time depending on your Internet connection and the package size.
Publishing...
The package has been accepted. We will notify you on email when process it.

C:\workspace\wolfssl-gojimmypi\IDE\PlatformIO>pio pkg publish --owner wolfssl-staging C:\workspace\wolfssl-gojimmypi\IDE\PlatformIO\PlatformIO_wolfSSL
Preparing a package...
Type:     library
Owner:    wolfssl-staging
Name:     wolfSSL
Version:  5.6.6-test1
Size:     10.75MB

Warning! A package with the name `wolfSSL` is already published by the next owners: wolfssl

Are you sure you want to publish the library wolfssl-staging/wolfSSL@5.6.6-test1 to the registry?
 [y/N]: y
The package publishing may take some time depending on your Internet connection and the package size.
Publishing...
The package has been accepted. We will notify you on email when process it.
```

See also my [PlatformIO Notes](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix/IDE/PlatformIO) that will eventually be found in the upstream [wolfssl/IDE/PlatformIO](https://github.com/wolfssl/wolfssl/tree/master/IDE/PlatformIO) (that's just 404 today).

Thanks again for your help.