platformio/platformio-registry #87: Auto fetch doesn't regards latest version code

ayushsharma82, 2024-02-28:
Hi @ivankravets ,

The registry's auto fetch program has a bug where it uses the latest `push` to repo rather than using releases & tags. Often the version code can jump around due to human-error which can cause the following issue with PIO registry.

You'll see the version code was first pushed by mistake as v4.1.0 to repo but later it was changed to v4.0.4. I hope your team can  fix the fetching script on your end and remove the rogue version.

[image]

Ref: https://github.com/ayushsharma82/ESP-DASH/issues/208

ivankravets, 2024-02-28:
To avoid similar issues, please use https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_publish.html

If you need to remove the broken versions, please respond to this ticket with the names/numbers that have to be removed.

ayushsharma82, 2024-03-01:
@ivankravets  Please remove `v4.1.0` of `ayushsharma82/ESP-DASH`. I'll take note of cmd_publish for next time.

ivankravets, 2024-03-01:
Fixed in https://registry.platformio.org/libraries/ayushsharma82/ESP-DASH/versions

closed 2024-03-01 by ivankravets