platformio/platformio-registry #90: Old versions have disappeared

nomis, 2024-03-17:
All of the older versions of my libraries have disappeared, leaving only the latest version:
* https://registry.platformio.org/libraries/nomis/uuid-common/versions
* https://registry.platformio.org/libraries/nomis/uuid-log/versions
* https://registry.platformio.org/libraries/nomis/uuid-console/versions
* https://registry.platformio.org/libraries/nomis/uuid-syslog/versions
* https://registry.platformio.org/libraries/nomis/uuid-telnet/versions
* https://registry.platformio.org/libraries/nomis/uuid-modbus/versions

This breaks all builds that happen to be using a previous version, and any attempts to build with an earlier version for debugging/bug fixing. What's the point of semantic versioning if you're going to unpublish all the older versions and break everything?

```
Library Manager: Installing nomis/uuid-log @ ^2.2.0
Error: Could not find the package with 'nomis/uuid-log @ ^2.2.0' requirements for your system 'linux_x86_64'
```

ivankravets, 2024-03-17:
Sorry, this a known issue. We have already applied the actions to avoid similar issues in future. 

See https://community.platformio.org/t/missing-packages-in-registry/39076/5

closed 2024-03-17 by ivankravets

nomis, 2024-03-19:
You've published a newer version of [tool-scons](https://registry.platformio.org/tools/platformio/tool-scons/versions) yesterday and now the older version has gone missing:
```
Error: Could not find the package with 'platformio/tool-scons @ ~4.40400.0' requirements for your system 'linux_x86_64'
```

nomis, 2024-03-19:
v6.1.5 is not 2 years old, and #4531 hasn't been fixed yet

nomis, 2024-03-24:
Is this going to be republished or not?

I don't like that you're making all the older versions of my projects impossible to build, especially when they can only be built with older versions of PlatformIO because of all the incompatible changes in v4, v5, v6.

nomis, 2024-03-24:
I have to use "ancient buggy releases" because the bugs don't affect me and they're the versions that work, whereas newer versions have different incompatibilities.

I can't upgrade because of platformio/platformio-core#4884 which I'm only now aware of because I had to downgrade to v6.1.5 when v6.1.6 came out for platformio/platformio-core#4531.

ivankravets, 2024-03-25:
> Could not find the package with 'platformio/tool-scons @ ~4.40400.0' 

This version is reverted to the registry. Please re-test your project with the latest PlatformIO Core 6.1.x and report all issues. We work on the PlatormIO Core 7.0 and there will be no support for 6.x soon.