platformio/platformio-registry #94: Incorrect owner, repo address, possibly cyber-squatting, please take down ASAP

tyeth, 2024-04-12:
## Report malware or abuse

Please fill out the form to report malware or abuse in a package to the PlatformIO team.

The PlatformIO team will review the report and act accordingly. In cases where malware or abuse is confirmed, the package will be removed and the namespace will be blocked.

Sensitive information should be sent to contact@platformio.org, included this issue link.

Please note that this report goes to the PlatformIO team for triage, not the package maintainer. To report bugs or vulnerabilities in a package please contact the maintainer directly.

#### Package information

- Package name: nthnn/Adafruit BusIO
- Package version: All
- Package link: https://registry.platformio.org/libraries/nthnn/Adafruit%20BusIO

#### Description of vulnerability

This has probably been accidentally published, but it is taking some of the adafruit BusIO installs, effectively cyber-squatting the library name. There is no listing on the official arduino library registry, and there is no public fork so this is not an "alternative" implementation or anything.

Just seen this warning today, but could have been there for up to 19days (when second library was published)
[image]

tyeth, 2024-04-12:
Just to add this one also looks garbage checking the github link, and presumably came from Seeed studios fork:
https://registry.platformio.org/libraries/nthnn/SIM900

ivankravets, 2024-04-17:
You need to use a valid library declaration as recommended at https://registry.platformio.org/libraries/adafruit/Adafruit%20BusIO/installation

Nevertheless, the duplicated library has been removed.

closed 2024-04-17 by ivankravets