platformio/platformio-registry #102: Claiming namespace AsyncElegantOTA

ayushsharma82, 2024-07-06:
#### Package information

- Package link: https://github.com/ayushsharma82/AsyncElegantOTA

#### Description

I am the author. 

Also, a question - how can I remove this library from PIO registry? It's deprecated.

ivankravets, 2024-07-06:
You can remove the library within 72 hours after its publishing. The library that you mentioned is trendy, with over 850 new installs for the last month. See https://registry.platformio.org/libraries/ayushsharma82/AsyncElegantOTA/insights/analytics

Once we remove it from the registry, multiple projects will be affected.

I recommend creating a new patch version (for example, `2.2.9`) and including a compilation warning that this library is deprecated.

ayushsharma82, 2024-07-06:
@ivankravets , Thanks for the reply. I know, **that's exactly why I want to remove the library from PIO registry** (it's already under process for arduino-registry), people should migrate to `ElegantOTA`.

Being the author of both, AsyncElegantOTA was marked deprecated long ago therefore It's not in my best interest to keep this up in registry. There is already a deprecation notice for it but people are following old tutorials and soon the library will break with platform changes.

ivankravets, 2024-07-06:
> It's not in my best interest to keep this up in registry.

This is exactly the purpose of the registry - to KEEP ALL PACKAGES and VERSIONS even if they are deprecated. 

I understand your position. However, it would be best if you thought from the end-user perspective. Usually, if something works, no one wants to change it. It would be best if you let our users have a choice of which version of the library they can use. As I mentioned before, publish a new version with a WARNING block to the deprecated library, and users will decide when they will have time to refactor their code base and migrate to the new version.

ayushsharma82, 2024-07-06:
@ivankravets Already done that, warning block was in v2.2.8 for past 10 months, that’s enough time for anyone to acknowledge it.

So as a library author, I’m telling you to remove the library from your registry. This is not a stranger who is asking you to remove a random library. I hope you acknowledge my request.

ivankravets, 2024-07-08:
The library has been deleted.

closed 2024-07-08 by ivankravets