platformio/platformio-registry #105: ESP-DASH licence violation and potential malicious code distribution,

dkaukov, 2024-08-14:
## Report malware or abuse

Please fill out the form to report malware or abuse in a package to the PlatformIO team.

The PlatformIO team will review the report and act accordingly. In cases where malware or abuse is confirmed, the package will be removed and the namespace will be blocked.

Sensitive information should be sent to contact@platformio.org, included this issue link.

Please note that this report goes to the PlatformIO team for triage, not the package maintainer. To report bugs or vulnerabilities in a package please contact the maintainer directly.

#### Package information

- Package name: ESP-DASH
- Package version: Staring from version 4.0.0
- Package link: https://registry.platformio.org/libraries/ayushsharma82/ESP-DASH

#### Description of vulnerability

I am writing to report a potential violation of the GNU General Public License v3.0 (GPL-3.0) in the project "ESP-DASH" by Ayush Sharma, which is hosted on your PlatformIO Registry.

**Violation Description**:
The project includes a file named `dash_webpage.cpp` which is distributed as a binary blob without providing the corresponding source code, as required by the GPL-3.0 license. I have attempted to contact the author via GitHub, as documented in [Issue #233](https://github.com/ayushsharma82/ESP-DASH/issues/233), but the author refused to comply with the GPL-3.0 requirements.

**Evidence**:
- [dash_webpage.cpp](https://github.com/ayushsharma82/ESP-DASH/blob/master/src/dash_webpage.cpp)
- [LICENSE](https://github.com/ayushsharma82/ESP-DASH/blob/master/LICENSE)
- [Issue #233](https://github.com/ayushsharma82/ESP-DASH/issues/233)

**Request**:
I respectfully request that PlatformIO review this issue and consider removing the library from the repository until we can be fully confident that the binary blob does not contain any vulnerabilities or malicious code.

ayushsharma82, 2024-08-14:
We have already gone through this issue in my repository. https://github.com/ayushsharma82/ESP-DASH/issues/233 . 
Your actions appear to be envious, and it seems that you do not desire others to benefit from positive outcomes.

To further clarify: the “binary blob” which you are reporting as “malicious” is just the ESP-DASH’s webpage gzipped and encoded in bytes format. So nothing really is obfuscated; if you had a little brain, you would have figured it out yourself and inspected it yourself _OR_ you could have got it inspected via a independent agency.

I have no interest in continuing this conversation as the whole thing is baseless and more or less a personal attack which happened on my 3 most popular libraries.

dkaukov, 2024-08-14:
Thank you for your response and for explaining that the binary blob is a gzipped and byte-encoded version of the ESP-DASH webpage. I appreciate the clarification.

However, I noticed that the gzipped webpage contains minified and obfuscated JavaScript code. According to the GPL-3.0 license, the source code should be provided in the form that is most suitable for modification—typically, this would be the unminified and unobfuscated version of the JavaScript.

My concern is that without access to the original, human-readable JavaScript, users may find it difficult to inspect, modify, or extend the code as the GPL-3.0 intends. Providing the unminified source code would help ensure full compliance with the GPL-3.0 and uphold the principles of open-source software.

I hope you understand that my intention is to support the integrity of open-source licensing and to help ensure that the community can benefit fully from your work. I look forward to your cooperation in resolving this matter.

ayushsharma82, 2024-08-14:
@dkaukov If you are not happy with it, please create your own projects and license them accordingly.

PS: You are talking to a law graduate so nobody can understand legal language better than me in this discussion. The GPL-3.0 license is applied to “licensee” (you) not the “licensor” which is me. Don’t make outcomes of stuff which you just read on ChatGPT.

dkaukov, 2024-08-14:
@ayushsharma82, Thank you for your response. I respect your background in law and your expertise in legal matters.

However, I believe there might be a misunderstanding regarding the application of the GPL-3.0 license. The GPL-3.0 is indeed a license that applies to both the licensor (the person or entity distributing the software) and the licensee (those who receive or use the software). As the licensor, when you distribute software under GPL-3.0, you agree to certain obligations, such as providing the source code in a form suitable for modification.

My concern, as stated earlier, is about ensuring that the project fully complies with these obligations, particularly regarding the provision of unminified and unobfuscated source code. This is not about questioning your intentions or expertise but rather about ensuring that the principles of the GPL-3.0 are upheld for the benefit of all users and contributors.

ayushsharma82, 2024-08-14:
You can keep on pursuing this baseless fight. I’m not participating in it no matter what’s the decision of PIO.

Thanks for wasting everyone’s time. Unsubscribing from this thread.

mathieucarbou, 2024-08-14:
FYI, for what is worth, as a owner of the commercial version,  and as a **heavy** contributor of ESP-DASH, having access to the source code of both Pro and OSS version, I can confirm that there is no malicious code or license violation regarding the used libraries in ESP-DASH behind or even abuse.

I wanted to clear this part.

Regarding the fact whether an author of a project licensed with GPL has to provide the source code of the "obfuscated" parts provided in the project in the sources, I am not a lawyer so I won't comment on it.

ayushsharma82, 2024-08-14:
@mathieucarbou This is a clear ego play from him because he was denied sources of proprietary webpage. Distribution of open-source software doesn’t mean any person will twist arm of the author at their will.

GPL-3.0 was specifically chosen for ESP-DASH and many of my other libraries because it allows for use of these “lite” versions in open-source but commercially available products while still allowing collaboration. What he is essentially forcing me to do is to switch to a BSL-1.1 license which is an copyleft license and I don’t want to do that because it will restrict everything to non-compete & non-commercial usage.

dkaukov, 2024-08-14:
At this point, I believe the best course of action is to escalate the matter to the GNU Project. However, I'm also concerned that this behavior might not only constitute a GPL-3.0 violation but could also represent an abuse of the PlatformIO distribution channel. It has the potential to mislead users and undermine the core principles of open-source software distribution. Additionally, I believe the security risks involved are significant.

ayushsharma82, 2024-08-14:
Go ahead. Make sure to get a signed letter with appostille stamp from them. You have no authority to comment on something/someone until you have a solid proof. Thank you.

ayushsharma82, 2024-08-14:
Since 12 years in open source, I’ve seen a lot of leeches like you who are happy to fight about just anything senseless. If I’m providing something for free in “open-source” that doesn’t mean I’m obliged to do anything. It’s my good-will that the project is open source under whatever license so that people can benefit from it regardless of what is being provided. It’s solely the decision of people if they want to use it or not.

So far the community has been very supportive and they understand the value that I bring with my open-source projects even if the UI part is not disclosed. They understand that the project has a paid component because without it, so called “open-source” is NOT sustainable.

Hopefully you will understand this one day and stop pursuing aimless issues. Your decisions will only ruin “open-source” further where stringent licensing will be necessary in edge cases like these. 

Over and out.

mairas, 2024-08-15:
I just happened to stumble on this discussion by chance and I have nothing to do with the issue or the software in question. 

Opening the upstream issue provided by @dkaukov and reading the issue and the first few comments, it is obvious that the source code includes compiled binaries without the respective source code, and this is done on purpose. I do not believe this is a copyright infringement, provided that the distributor is also the copyright holder of the UI. They are free to distribute the source code and the binaries under any license of their choosing.

However, it does seem to me that the ESP-DASH source distribution indeed does not follow the letter or the spirit of GPL. This becomes an issue whenever anyone attempts to distribute a derivative work, or worse, uses some other GPL v3 library together with ESP-DASH. Given that the upstream does not provide the required source code, it would be impossible for the derivative work author to be in compliance of the license. This is also the reason why exceptions to the GPL terms can't be made. 

Given that the ESP-DASH project is also the copyright holder of the source code, it is trivially simple for them to become compliant: either distribute ESP-DASH under a non-copyleft license, or provide the missing source code. If neither of this is done, I suggest the offending library is removed from the PlatformIO registry.

As a sidenote, I do find @ayushsharma82's communication style and personal insults off-putting and unprofessional. You can do better.

As a second sidenote, @dkaukov might want to check GitHub's ToS related to such licensing discrepancies. And no, I don't think reporting these issues is petty, it's about ensuring a level playing field for anyone working in open source. Like myself.

mathieucarbou, 2024-08-15:
@ayushsharma82 : to avoid any fuzzy interpretation or grey area, a solution would be:

1. to publish the .gz output file of the web part (where the sources are not available) in the release section of the project, as an artefact. A program output is not covered by any license as stated in the GPL FAQ, so users could download this .gz file

2. Remove the embbeded binary from the headers and provide some documentation to explain how users can download the .gz file, put it in the data folder, and upload it in their file system partition. 

The project will be split in 2 parts

1. The GPL code with the c++ sources (where the web handler would server a static gzip file from fiel system instead of embedded data)
2. The output of the web layer as gzip, published in the release page. Up to the user to include or not this blog in his project.

The blob is then not linked, neither part of the project anymore, it comes as an artefact served from the file system.

mairas, 2024-08-15:
@mathieucarbou I believe this would be a totally workable solution (referring to the comment version at 10:13 UTC). You would be doing two distributions, one under GPL v3 and one under any other license of your choosing.

Semi-related (but does not affect this solution): The `.gz` file is compiled output (object code, in GPL v3 parlance) of the HTML and JS sources and thus definitely under the scope of the license. The program output FAQ section refers to whether, say, a text file written using a GPL-licensed text editor would need to be licensed under GPL or not. But this does indeed not affect your proposal, which is fine in my opinion.

mathieucarbou, 2024-08-15:
@mairas : I hope @ayushsharma82 will consider one of these solution too because it will also help for this issue I've opened (https://github.com/ayushsharma82/ESP-DASH/issues/217) about reducing firmware size and placing the generated output as a gzip file in a partition.

ayushsharma82, 2024-08-15:
@mathieucarbou  @mairas  I respect your kind opinions. Yes, my behaviour might be slightly unprofessional in some comments but If you look at it from my viewpoint and a person who is just after you for 12 hours straight senselessly, It only takes a few minutes to snap. This could have been handled much better if dkaukov approached me in a different way.

In any case, I'm working on a BUSL license with additional use grant that will retain same privileges as GPL for users.

dkaukov, 2024-08-15:
Hi, also, I believe,  if the .gz file is built using OSS or includes OSS-derived content, a license notice should be provided. This notice helps ensure compliance with the original licenses and informs users of their rights and obligations regarding the file. Properly documenting the license in both the repository and within the .gz file itself (if possible) would be best practice.

dkaukov, 2024-08-15:
yeah, there are bunch of projects affected @ayushsharma82 @mathieucarbou   do you want me to create separate tickets for each of them?

ayushsharma82, 2024-08-15:
@dkaukov Do you have a signed letter of violation from GNU? You sure can once you have a proof of violation but right now this just based on your assumptions. 

The relicensing is being done as a precautionary measure.

mairas, 2024-08-15:
> @dkaukov Do you have a signed letter of violation from GNU? You sure can once you have a proof of violation but right now this just based on your assumptions.

The Free Software Foundation really isn't a stakeholder here; they might provide guidance because they have designed the license. The parties would be the ESP-DASH project (or rather, anyone who happens to own the copyright), potential developers who might distribute derivative works, as well as any intermediaries such as GitHub and PlatformIO. I don't think there would be any limitation for you to distribute software under conflicting or impossible license terms because the only injured party would be you. However, any downstream distributors would face actual reputational or financial risk.

dkaukov, 2024-08-15:
Ayush,

I've noticed that several of your projects, including [NetWizard](https://github.com/ayushsharma82/NetWizard/blob/main/src/nwp.cpp), [ElegantOTA](https://github.com/ayushsharma82/ElegantOTA/blob/master/src/elop.cpp), and [WebSerial](https://github.com/ayushsharma82/WebSerial/blob/master/src/wslp.cpp), use a similar technique of embedding binary blobs without providing the corresponding source code. I believe these projects may face the same licensing and distribution issues we're discussing with ESP-DASH.

My question is: Do you plan to address these issues across all of these projects, or would you prefer that I create separate issues for each one?

mathieucarbou, 2024-08-15:
@mairas : in my case (as a developer heavily using @ayushsharma82 projects), and then consequently distributing them under GPL, I would be more concerned about me not complying with the GPL license...

From the discussion above I understand that since the source code of the web part is not made available, I then cannot even include it in my projects, even if GPL is used to comply with @ayushsharma82 's project license ?

That's the part I don't understand because the distribution is done in GPL, on GitHub, public repo, my source code is available, and consequently those of the dependencies... From my point of view as a user of the project, I am  considering it as sourde doe whether it include binary blob or not. 

If the library I am depending on is a font library, which is heavily using embedded blobs, should I make sure that this project also have the sources and l mechanism to create these artefact ? Personnally, I don't mind. 

What I am interested is in the correct use of a library covered under a specific license. As long as my source code of my project is made available under GPL, I should not have to go through the diligence work of exploring and making sure that all the libraries I use behind are all providing source code for every part they provide?

mathieucarbou, 2024-08-15:
@mairas yes, I understand. Thanks!
So yes this is a problem indeed...

ayushsharma82, 2024-08-15:
### Final Statement 

In light of the assumptions made by dkaukov + others involved in this discussion, and specially due to ambiguity in certain elements of the GPL-3.0 license, I have made the decision to retain the current licenses for my libraries (ESP-DASH, ElegantOTA, Netwizard and WebSerial). My legal reasoning can be interpreted as follows:

### Nature of Code:
Since the webpage content is stored as an array of bytes within a source file, it is essentially a small part of my program's source code. From the perspective of the GPLv3.0, this byte array is treated as part of the library's source code, not as a distinct piece of content.

#### Source Code Definition
The GPLv3.0 defines "source code" as the preferred form of the work for making modifications. Since it is embedded within the source code, anyone receiving the program under the GPLv3.0 has the right to modify that byte array (even though it might be less convenient than modifying the original HTML) and redistribute the modified version.

### No Obligation to Disclose Original Content:
The GPLv3.0 does not require me or the licensees to disclose the original content that generated the byte array because the original content was never part of the distributed library. The byte array itself, as part of the source code, suffices as long as it is the form used in the program.

---

**No further feedback by anyone will be entertained. We have already wasted a lot of time.** Licenses are a sensitive topic and should have been discussed professionally & legally in private by both parties before resorting to public outcry by the accuser. As the copyright owner, I will continue to provide these libraries under the same license stating my reasoning above.

If anyone still has concerns regarding the violation of the license, they are welcome to initiate legal proceedings within my Court of jurisdiction. This matter is not subject to community discussion.

Regarding the claims of the file being "malicious", you already have @mathieucarbou who has gone through the source, there is nothing malicious in my libraries and contain MIT dependencies. If anyone is still suspicious, they can get it inspected via an independent authority and report back with a proof.

mairas, 2024-08-16:
I don't believe anyone is claiming Ayush is not fully within his rights to distribute his software under any license he wants, even if the content is in conflict with the license. However, since that is the case here (a binary blob is _not_ the preferred form of work, just like Ayush admitted), the license discrepancy should prevent all third parties, including PlatformIO and GitHub, from distributing that code under GPL v3.

I also find it a bit silly that Ayush starts waving the prospect of legal action due to this issue. As I said before, you can do better. There is no injured party, and Ayush is fully within his rights to distribute content he owns the copyright to as he wants. At the same time, there should be a clear understanding that no other party should be obliged to further distribute code with invalid licensing.

ayushsharma82, 2024-08-16:
@mairas  Look, I don’t have energy and time left to be paranoid about open source licensing of projects mostly used for hobbyists, students and small businesses so I don’t care if anyone removes the library or not and you agreed there is no injured party so why are we here? Did this need a public outcry?

If you are a person who is “tell everyone” type, just admit it, message these platforms privately and let them decide on their capacity.

While engaging in a public conversation with speculations, you are trying to defame me and my projects. Which I guess was the motive of @dkaukov due to his persistent nature of comments.

You are not part of PIO, Github or any other platform where the library is “hosted” so commenting here with “interpreted” thoughts about a license is waste of time for everyone.

I hope there is no need to talk more now.

mairas, 2024-08-16:
There is significant value for having accurate metadata in software repositories such as PlatformIO or GitHub. If someone makes development decisions based on that metadata, and it turns out to be inaccurate, that can become very expensive, both in terms of time and money. I understand it is in your business interests to avoid fixing the licensing issues, but that still doesn't make it right.

Again, your commentary of what type of a person someone is, or whether something is waste of time or what someone's motives are, is unwarranted and unprofessional. You can do better. Focus on the substance.

ayushsharma82, 2024-08-16:
Good. End of discussion now. I don’t want any further comments on this topic. I hope @dkaukov & @mairas  learn to respect that.

mairas, 2024-08-16:
Note that this topic is primarily about the software being distributed in PlatformIO with incorrect licensing. It is for PlatformIO staff to resolve this issue and decide when the discussion is done.

ayushsharma82, 2024-08-16:
Here you go, added "written offer of source code" as per section 6 if anyone 'thinks' dash_webpage.cpp is considered an object code. This puts all my libraries in line with GPL license and there is no ambiguity left to argue about.

https://github.com/ayushsharma82/ESP-DASH/blob/master/docs/Written%20Offer%20for%20Source%20Code.pdf

mairas, 2024-08-16:
@ayushsharma82 thank you, I appreciate your admission that you need to provide the source code for the binary blobs embedded in the source code! This has moved the discussion ahead in a significant manner!

I believe there might have been a tiny confusion, though. You seem to be quoting in your offer GPL v3 section 6.b. Section 6 refers to alternative methods of conveying the object code and states the minimum requirements for conveying the corresponding source code. The section you quoted applies to object code being conveyed embedded in a physical device. Since the object code, in reality, is placed available according to section 6.d, you are required to "offer equivalent access to the Corresponding Source in the same way through the same place at no further charge." I hope this clarifies the license requirements!

Furthermore, even if your proposed approach would have been valid, that would in practice have blocked any redistribution of your libraries or any derivative works, unless the distributor first acquired the source code from you.

ayushsharma82, 2024-08-16:
@mairas I have not admitted to anything. This is entirely based on presumptions any other person would have regarding what’s called a binary.

Orcale and other companies all do this with GPL software which they distribute so this is no surprise that it won’t apply here as well.

GPL is very vague around these terms and people can make anything out of it. If you are not happy with the solution, it’s advised that you hire a attorney and precent your facts.

@ivankravets The best you can do is lock down this issue because these people won’t stop with any solution and just  want to defame the project with a agenda. And most importantly it should something that your team discusses privately.

mairas, 2024-08-16:
What you are referring to with Oracle and other companies applies to GPL v2. This requirement has been updated in GPL v3, and if you read the license, the language is understandable and unambiguous. I warmly recommend doing so!

Again, you seem to be making assumptions on my motives and agenda, even though the issue at hand is quite simple and could easily be solved by you. It is not very constructive to demand the discussion be stopped instead of fixing the underlying issue.

ayushsharma82, 2024-08-16:
See this is what I'm talking about, you guys just don't shut up regardless of what answers or solutions I provide, which can make people lose their professional sense. I agree that you have stated your point and I've acknowledged it in earlier comments but who are you? Are you the injured party? Are you in team of the distributors? If yes, I would have happily listened to you.

You are neither of them so don't cast your opinions on something which the other person has to think about. This just explains your character more than anything. You are trying to be an obstacle in my business interests and defaming me talking "license" as your agenda. If you have so much interest to interfering with anybody's business then this is not the right place to be in.

I'm agitated because this should be interpreted by PIO themselves and what they make out of the license. We both have stated our facts and solutions, but your opinions and assumptions influence the decision. Do you want to be help liable for the actions? If so, continue rambling.

mairas, 2024-08-16:
I understand that you are unhappy about the situation, and I understand how it might be upsetting to have to fix it, but the license issue is real and a problem for both PlatformIO registry's trustworthiness and for any developer using your code without knowing about the license conflict. It doesn't really matter who has reported the issue - it persists until it is addressed, and it isn't wrong to expect a level and fair playing field among open source developers.

I see at least four ways to resolve the matter:

1. The offending binary blobs are removed from the GPLed codebases.
2. The source codes of the binary blobs are released.
3. The projects are re-licensed.
4. The projects are removed from PlatformIO.

None of these options are complex and all require less work than the mailing scheme you attempted to set up. As I've said multiple times before: You can do better!

dkaukov, 2024-08-16:
@ayushsharma82, perhaps it might be worth considering selling and distributing proprietary software through the PlatformIO registry's [commercial offerings](https://registry.platformio.org/pricing). This could be a fairer approach that aligns better with the current setup and avoids any potential conflicts with open-source licensing. Additionally, making your repositories private could also help manage the distribution of your work in a way that suits your business interests.

closed 2024-08-17 by ivankravets