platformio/platformio-registry #111: Claiming namespace [Witty-Wizard]

Witty-Wizard, 2024-11-28:
#### Package information

- Package link: https://github.com/Witty-Wizard/SerialIO

#### Description

I have created this library you can check this on my github history of the repo I even have it published on the Arduino Library Manager However, PlatformIO library manager does not seam to pull the changes from the arduino library manager

ivankravets, 2024-12-21:
>  PlatformIO library manager does not seam to pull the changes
Because you publish the latest version manually. Once you do this, the PlatformIO Registry disables automatic updates. So, we reenabled this feature again for that library. https://registry.platformio.org/libraries/witty-wizard/SerialIO

closed 2024-12-21 by ivankravets

Witty-Wizard, 2024-12-21:
Ahh, I see.
So from now on if i publish a new release it should update automatically right?

ivankravets, 2024-12-21:
If you mean [this](https://github.com/Witty-Wizard/SerialIO/releases), then YES. Just do not forget to update `version` field in the manfiest.